westlicht/performer #198: Notes in sequence transposing randomly when changing scales

captainhooke88, 2020-05-30:
Hi,

I have some unwanted behavior it seems like when I’m changing scales in a sequence or in the project.

It seems to transpose certain steps only and in most cases they are transposed up around 3 octaves seems like..

Anything I’m missing here perhaps?

westlicht, 2020-05-31:
Yes, this is because note numbers are translated to CV using the concept of voltage tables. An _octave_ contains as many notes as the voltage table contains. The _semitone scale_ for example has 12 entries, whereas _minor_ or _major_ only have 7 entries. So if you enter a sequence while using the _semitone_ scale, it will not sound the same when switching to _major_, as you might expect from other quantizers that just snap notes to the nearest one in the scale. I know that might be a small negative in some cases, but the rule is to stick with scales that have the same number of notes in there (or choose one and then enter notes). The big advantage is that you can have scales with more than 12 entries (see the _24-tet_ scale for example) and if you transpose a sequence, it will actually move every note to a new note other than with a quantizer that might still select the same note after transposition. It's just a different concept to get used to.

captainhooke88, 2020-05-31:
Alrighty good to know. Definitely something to get used to for me personally since I’m used to more “traditional” quantizers. It does imo also make cv controlling scale changes a bit hard to work with.  But I did hear some gossip about some harmony manager like track setting to come up with interesting progressions possibly in the works... 

Anyway good to know how it works and that nothings wrong with my unit. Thanks very much for explaining and all the hard work!

westlicht, 2020-06-01:
You're welcome!

closed 2020-06-01 by westlicht