westlicht/performer #199: Random constraints

theotteryears, 2020-05-31:
I was messing around with the CV in to change the division of a track yesterday, and wondered if its possible to introduce some kind of screen for constraining the choices here, so you could choose which divisions you wanted to be in the selection the CV input chose from.

At the moment, the timing can get really wild with all the in-between musical divisions.

SSF Gatestorm has this random constraint system and its very useful (in case what I wrote was unclear). thanks :)

westlicht, 2020-05-31:
I remember back when routing for divisor was requested as a feature I was sceptical because of that exact reason. Without selecting a set of possible divisors, results will be extremely random. The requester didn't mind so I added it. Constraining to a set of values is a bit tricky, but I might give it a bit more thought.

theotteryears, 2020-05-31:
Ok great! Yeah it doesn’t make much musical sense at the moment unless you control it from another sequencer so you can dial in exact values. Maybe there’s an option for another page where you can select from which divisors you want to include etc.

westlicht, 2020-06-14:
This is low priority, moving to backlog.

closed 2020-06-14 by westlicht