westlicht/performer #201: Micro-timing

theotteryears, 2020-06-03:
I have a Circadian Rhythm as theres a good feature where you can zoom into a step and see the micro timing grid (24ppqn). is it possible to have something like this added to your sequencer?

Maybe a new page which switches to the micro timing grid. 24ppqn would be certainly adequate for fast hi-hat things.

westlicht, 2020-06-06:
There is the _Gate Offset_ layer that allows you to do micro-timing. Zooming in would require there to be more step information, which is not possible due to memory constraints.

theotteryears, 2020-06-06:
Yes it was the zoom that I wanted rather than shifting steps using the offset. 

Like zoom into a 16th beat to see the underlying 24ppqn for faster rolls etc. Shame there’s no memory for this but thanks for the reply.

theotteryears, 2020-06-06:
Could you just then add a 6 beat retrigger to the retire page if possible? I only need 6, 4, 3 and 2 retrievers - nothing more complicated so it would only need a small addition and not the zoom feature like I originally thought.

theotteryears, 2020-06-10:
Actually its the CURVE track type that's the best for drum programming - as you can vary the gates much better than using a retrigger on the NOTE track. So I am rephrasing my request... could we please have up to 6 notes on the gate page of the Curve track type - with all the in between options like there are already. this would allow triplet programming.

westlicht, 2020-06-14:
I might increase the number of retriggers to 8.

noahsider, 2020-07-06:
would love to also be able to control the timing of retrigs for glitchy mayhem. One of my fav features on the octatrack.