westlicht/performer #202: Grouping events

theotteryears, 2020-06-03:
Could a new option be added to the CONDITION page where you can group events together into a group and then have this group trigger or not (all together)?

Eg. I want a group of 4 x 16th notes to either all play or all not play.

westlicht, 2020-06-06:
That's what the `Pre` condition is for. Set the first of the 4 notes to whatever condition you want the group to trigger for. Then use `Pre` on the next 3 notes. The 4 notes will now either all play, or not play. I assume that should be exactly what you want. Using `!Pre` together with `Pre` you can even have either steps 1 and 3 play, or 2 and 4 or any other combination. This seems more flexible than grouping events.

theotteryears, 2020-06-06:
Perfect. Sorry - I didn’t realise that was what the PRE condition did.

closed 2020-06-06 by westlicht

theotteryears, 2020-06-08:
I can't get this to work properly. 

I have a gate set to trigger 50% of the time, and 3 notes after it set to PRE. these fire regardless of whether the step before it triggers or not.

What I would like is for all these 4 steps to be tied together so that when the first one plays, the rest should also play. From your description above, what I am trying should work, but it doesn't.