westlicht/performer #203: Improve handling of fills

theotteryears, 2020-06-05:
Try tapping quarter notes or 16th notes out on one of the fill buttons - half of the time the button is not responsive enough (or is quantising) or just completely missing sending the trigger out, though the button does light up. can this be looked at as it's making live performing fills a little hit and miss at the moment :)

westlicht, 2020-06-06:
The fill buttons are not intended to trigger individual gates, but to _activate_ the selected fill mode on that track (there are multiple modes). It might be interesting to add a new mode that allows to trigger gates directly, but the current implementation is doing exactly what it's supposed to in my opinion.

theotteryears, 2020-06-06:
Ok no worries :) just thought I’d ask. Maybe I shouldn’t be using it like I was!