westlicht/performer #204: Override of CV pattern change from pattern screen?

theotteryears, 2020-06-07:
Would it be possible to add an override of the current pattern when its set to change via CV input, from the pattern screen?

e.g. I am changing the patterns for all tracks from CV input 1 from another sequencer. however, I want a single channel on Performer to be able to override this so I can bring other patterns in and out. I would exclude this from the CV input usually, but I want it to follow to main arrangement most of the time and just let me pick and choose when to deviate from this.

westlicht, 2020-06-14:
What you are basically asking for is an override of an override (cv input being the first override). This is not something that makes a lot of sense in general and I like to keep the routing system generic and uniform.

I'd suggest to use two external cv inputs. Use an external switch to control if the second cv is a copy of the first or from another source like a DC fader.

closed 2020-06-14 by westlicht

theotteryears, 2020-06-14:
Yes the reason for this is that I’m arranging the Performer patterns from another sequencer. This forms the main arrangement. Then I was hoping to change just certain drum patterns while playing live. 

Therefore needing the override of the override.