westlicht/performer #205: Make fills accessible from other screens

theotteryears, 2020-06-07:
It would be useful for the fills to be accessible from the pattern and overview screens. Seems like the buttons used for fills are unused on these pages... would be helpful as I'd like to watch the overview screen and perform fills at the same time!

westlicht, 2020-06-14:
Just hold performer key. Its the same on any page. Or do I miss something?

closed 2020-06-14 by westlicht

theotteryears, 2020-06-14:
I just asked so that you can actually watch the overview screen whilst performing fills without having to change screens. The buttons are unused on the overview screen at the moment.

westlicht, 2020-06-14:
Which you can by holding the performer page key and a fill key. This is something I do all the time (the same with mutes) and having your muscle memory do this two key combo allows you to do fills/mutes from whatever page you are. That was the initial concept of having the performer and pattern pages always accessible by holding the respective keys. I'm not a big fan of having multiple ways of achieving the same thing if not necessary. I hope you can see my point.

theotteryears, 2020-06-14:
Yes I see your point. Fair enough and thanks for all your responses! :)