westlicht/performer #207: Grouping events

theotteryears, 2020-06-08:
I can't get this to work properly. 

I have a gate set to trigger 50% of the time, and 3 notes after it set to PRE. these fire regardless of whether the step before it triggers or not.

What I would like is for all these 4 steps to be tied together so that when the first one plays, the rest should also play. From your description above, what I am trying should work, but it doesn't.

_Originally posted by @theotteryears in https://github.com/westlicht/performer/issues/202#issuecomment-640734654_

theotteryears, 2020-06-08:
Could you please make this so that the steps set to PRE will occur whether or not the leader of the group of notes has a condition set or not. This only works if there is a condition assigned to the first note of the "group'. I want the main note to hit 50% of the time the pattern plays-  and the notes after it to follow this. This only works if the main note is set to 1:2 but then it only occurs every second time through the pattern. The conditions tab could have a 1:1 option adding, which would get around this?

westlicht, 2020-06-14:
Yes, the solution is in #129 which I intend to fix. No need to do any grouping of events, just have gate probability also affect the pre condition.

closed 2020-06-14 by westlicht