westlicht/performer #209: Automatic power off after around 10 minutes of inactivity

Nek, 2020-06-09:
Hello,

Is it a normal behaviour? 
Is it the same on the latest firmware?

Thanks,
Nik

theotteryears, 2020-06-09:
I don't have this happening to mine - and I leave it on for hours

captainhooke88, 2020-06-09:
do you maybe mean that the screen turns off? not the whole unit right? I think its just to prevent burning in of the screen or something, doesn't bother me!

Nek, 2020-06-09:
@theotteryears Thanks! That's very good to know.
@captainhooke88 I'm not sure (it's from the other person's words) but it requires power reset to turn back on.

I'm asking because I'd like to buy one and the owner has told me he has this weird issue. He has presented it as something normal, a quirk of the older OS. He hasn't upgraded to the latest firmware. But I highly doubt it is normal behaviour.

captainhooke88, 2020-06-10:
Can not expect that's normal behaviour, but I have the latest firmware and do not have this problem at all and can leave it on for as ever long I want. The screen goes to sleep after some time, but no issue there.

Nek, 2020-06-10:
It looks like a defect. I'll look for another seller.
Thanks!

closed 2020-06-10 by Nek