westlicht/performer #212: Mutes accessible from Pattern screen

theotteryears, 2020-06-11:
Or patterns available from the mute screen as these 2 functions go hand in hand and it would be useful to not have to switch pages.

theotteryears, 2020-06-12:
eg. I'm triggering a bass drum fill as a transition to the next pattern... the two screens make this very difficult to accomplish.

westlicht, 2020-06-14:
You can use sync to prepare the pattern switch and then use mutes, I don't really see a problem. Mutes are available on any page just holding performer button. I don't plan to change this concept.

closed 2020-06-14 by westlicht

theotteryears, 2020-06-14:
The problem is you don’t have enough fingers to perform a fill and change patterns. 

I know that you can quick access these pages but triggering fills at the same time as changing patterns is quite hard. 

Basically I just thought pattern changing, mutes and fills should all be on the same page as that’s performing to me :)

westlicht, 2020-06-14:
Yes I can see that, but there is no easy way to get all of that functionality on a single page. I think the current separation makes a lot of sense, but feel free to propose a better variant without loosing the existing functionality.

theotteryears, 2020-06-14:
I think there’s space on the performer screen.

Instead of the middle of the square showing whether a track is muted, the pattern number could be displayed here.

The mutes could move to T1 - T8 numbers at the top… Either with a square or by a strikethrough of the track number. (Doesn’t matter too much as the LED goes red when muted anyway.)

Buttons S1 - S8 could be used for pattern selection (maybe with the use of shift? For patterns 9 -16)

The F5 key is free to use for FILL PERCENTAGE.

I’ll give it some more thought! 

It’s fine how it is, just this occurred to me when writing and performing a track the other day.