westlicht/performer #214: Buttons stop working in generator modes.

vdgno, 2020-07-07:
I have ver 0.1.35. when entering random or euclidian generator from step screen, all sequencer buttons stop to response except function buttons. Definitely a bug here.

westlicht, 2020-07-19:
Not really a bug, the generate page is a modal page, meaning it captures all input. What buttons would you have expected to keep working? Play/stop? Other?

vdgno, 2020-07-20:
At least start stop buttons and the ability to leave generative mode back to normal would be great.

westlicht, 2020-07-20:
You can leave the page by either using _COMMIT_ or _REVERT_ in the context menu (hold `SHIFT` + `PAGE` key).

westlicht, 2020-07-20:
But there is a reason the generators are not yet documented. It's a sort of unfinished feature for a long time now. See #12

westlicht, 2020-07-20:
Closing this, as it's not a bug per se.

closed 2020-07-20 by westlicht

vdgno, 2020-07-21:
> You can leave the page by either using _COMMIT_ or _REVERT_ in the context menu (hold `SHIFT` + `PAGE` key).

great. thank you