westlicht/performer #216: MIDI Monitoring behavior while sequencer is playing (possible bug?)

MDCusack, 2020-07-25:
Hi Simon,
firstly - just want to gush over how much I LOVE my Performer, you've done an incredible job! I've had it for about 6 months and I continue to find new tricks all the time!

I just have a query regarding incoming/outgoing MIDI signals. It's possible there's a setup parameter im missing, but the inconsistency makes me think this is a bug.

My setup is MIDI keyboard to MIDI input. MIDI output to Elektron Digitone. Configuring the MIDI output to MIDI (1) and Note events works fine while the sequencer is stopped - I can play my Digitone with the keyboard as you'd expect.  **When the sequencer is running though (with an empty track and recording off), MIDI notes that i play with my keyboard get stuck on, sometimes dont trigger and sometimes behave correctly.**  I've also tried this with another MIDI device and the behavior is identical.

Changing the CV update mode to from Gate to Always makes it behave even weirder, doing pulses of 4 notes at a time when a key is held. 

Once a sequence is recorded, it plays back correctly - no issue there.

This workflow works perfectly when using the CV outputs rather than the MIDI out, i'm just hoping to replicate that workflow over to my MIDI devices.

Any tips?
FYI i'm using the latest firmware.

thanks again!

napoleonthegreat94, 2020-08-09:
I experience the same behaviour. It would be great if that could be fixed.

westlicht, 2020-10-15:
Sounds like a bug yes, will have to investigate.

westlicht, 2020-10-22:
Fix will be released with 0.1.38

closed 2020-10-22 by westlicht

napoleonthegreat94, 2020-10-22:
thats great, thanks!

commit 03f8a076 (2020-10-22): fixed live monitoring and handle monitor modes

fixes #216