westlicht/performer #218: Bug: Curve track CV output turns off when recording is enabled

MDCusack, 2020-08-03:
Hi Simon,
firmware v0.1.36 seems to have a bug that turns the curve tracks off when record is enabled, looking at the monitor screen confirms this. I've reverted to v0.1.35 and it works correctly. 

No hyper specific steps are required to reproduce it, just make a curve track, put in some curves, hit play - works fine, hit record and the output stops outputting voltage until record is turned off :(

napoleonthegreat94, 2020-09-08:
@MDCusack i have the same experience. it is minor problem but can be annoying in some cases

westlicht, 2020-09-23:
@MDCusack I can't reproduce it here. Can you upload a project file for me to investigate more?

napoleonthegreat94, 2020-09-23:
if for example i route a curve track to one of the routable parametres (tempo for instance). when i hit record, the routable parameter does not follow the curve track anymore. hope i helped

westlicht, 2020-09-24:
Aha that makes more sense. Will look into it. Thanks for reporting!

MDCusack, 2020-09-24:
Hey guys, just re-installed the latest firmware and tried again. Confirming the behaviour I get is that curve tracks cease outputting voltage when record is enabled - they become a static -5v until recording is disabled. 

In this video I initialised the project, recorded some note data on track one, set track two to be a curve track, made it a single step track with time division slowed right down. Happens regardless of whether clock is master or slave.

https://photos.app.goo.gl/4P1VJ7fYY6Kn1wL27

MDCusack, 2020-09-24:
here's the project

[performer test.zip](https://github.com/westlicht/performer/files/5275836/performer.test.zip)

westlicht, 2020-09-24:
This is a duplicate of #189 and is already fixed. Will be part of 0.1.37 release.

closed 2020-09-24 by westlicht