westlicht/performer #219: New feature? destiny Routing: step forward

franq87, 2020-08-03:
Could it be possible this feature? routing a track that each step could be externally clocked so I can make a short of sample&hold and use that track like a quantizer?

MDCusack, 2020-08-03:
Weird, i was literally looking to see if that was possible last night. Adding the option to unlink a track from the main clock and only have it proceed steps forward upon receiving a 5v hit from a CV input setup via routings would be very cool.

franq87, 2020-08-04:
@MDCusack yeah It would be very cool, let's see if it's possible

napoleonthegreat94, 2020-08-09:
@MDCusack i would reaally like to see that too! but then, how would the song mode work with different clocks going on?

westlicht, 2020-10-15:
I can see the appeal of such a feature but the current architecture is not really suited to implement this and would mean ugly hacks. So I'm afraid this is not going to happen, sorry.

closed 2020-10-15 by westlicht