westlicht/performer #224: Auto-fill empty steps

Materia-Oscura, 2020-08-25:
Hi,

One feature i would really like to have is the hability to program 8/16/whatever steps, then having those steps repeated if the sequence lenght is expanded.

Lets say i set up a A/F/E/A/E/E/F/G sequence in 8 step "mode" and I  extend the sequence lenght to 32 steps, the sequencer should fill the empty steps with the same sequence of notes repeated over and over again for the 32 steps (A/F/E/A/E/E/F/G/ A/F/E/A/E/E/F/G/A/F/E/A/E/E/F/G/A/F/E/A/E/E/F/G) i can then edit this sequence further to make variations and such

Does it sound possible?

westlicht, 2020-08-25:
You can use the duplicate feature in the context menu for that. I do that all the time to get from a short to a long sequence and add details.

Materia-Oscura, 2020-08-25:
> You can use the duplicate feature in the context menu for that. I do that all the time to get from a short to a long sequence and add details.

Thanks for the answer!
Discovered this feature just before I see your post, it works well. Not as handy as an automatic filling when you got a 3 steps sequence and you want to fill 64 steps with it but still very usable ;)

Off topic but, do you still plan to support the APC mini with it's 8 faders to input notes steps?

westlicht, 2020-10-15:
I really like the duplicate feature and use it all the time. The problem by doing it automatically is that you loose control when to do it, which I don't like at all.

As for the APC mini. I don't have one, so it's not on my todo list. But someone could add it. This is why the firmware is open source!

closed 2020-10-15 by westlicht