westlicht/performer #226: track and output clock delay

Lokua, 2020-09-12:
**Feature request:** 

Per-track delay option to offset a track by N milliseconds, also available for the output clock.

This would help solve an issue for me that has nothing to do with the Performer, but with other digital modules in my rack that suffer from latency. Being as Performer is a "brain" type of sequencer, being able to solve this here would be the most elegant solution in my opinion, but of course I'm biased as this is a big problem for me.

**Some context to more fully illustrate the issue I have:**

I am using the Performer as my master clock via clock out.
I am using 4 tracks on the Performer to trigger 4 analog voices.
The remaining 4 tracks are going into a digital module that has 9ms latency (an ER-301, but pretty much all digital modules / samplers are going to have some latency).

To work around this issue I:

- added the minimum gate offset to all the tracks that are triggering analog voices (12.5% I believe it is)
- sacrificed an entire track with the same gate offset to act as master clock for the rest of the system (this is the one that hurts a lot)
- used the remaining 3 tracks _without_ a gate offset to trigger drums on the ER-301
- the minimum offset value actually pushes everything just a tiny bit too far from the non-offset triggers after they go through the 9ms latency, but fortunately the ER-301 has a micro-delay so I compensate with that (this is also dependent on tempo so I have to readjust)

So again being able to apply a constant, static millisecond offset to individual tracks as well as the clock out would be amazing as I would be able to keep everything super tight and wouldn't have to use a track for the output clock. Kind of edge request perhaps I really don't have an idea how big of an ask this is implementation wise, but thanks for your consideration.

dComposer, 2020-10-15:
I noticed this problem when sending clock out from performer to Grids. Grids just consistently seemed to be off by a few ms. Perhaps this could fix that issue?

theotteryears, 2020-11-23:
Grids (at least my grids) requires the reset sent to it to come slightly ahead of the clock (otherwise you get a weird stutter sometimes)