westlicht/performer #228: MIDI CV step recording options

natedey, 2020-09-17:
Hi there,
Recently got the Performer, and my goodness, what an incredible module!!!
I have a suggestion that might improve MIDI step sequencing
The Mutable Instruments Anushri desktop synth has a step sequencer that can recognize pitch bend and mod wheel movements to manipulate the step recording. On that box, pitch wheel movement (magnitude and direction don't matter) adds glide to the step, mod wheel movement adds an accent to the step. I always thought that was less useful than having rest / tie assigned to pitch and mod. 
Would it be possible to have hard-coded or assignable options to bend/mod/or other CC to be able to input a more nuanced sequence completely on the controller? Thanks!

westlicht, 2020-09-21:
Very good idea. It will be part of the 0.1.37 release.

natedey, 2020-09-21:
awesome! Thanks Simon

commit 07614f67 (2020-09-21): added step recording with slides/ties

fixes #228
closed 2020-09-21 by westlicht