westlicht/performer #230: Ability to init one layer data  only

shsCT, 2020-09-27:
Would love the ability to init say, just the note range layer or a probability layer, rather than globally initializing all step info at once. Thanks!

captainhooke88, 2020-09-27:
Just from the top of my head you might be able to do that with The generate function. Go to the function you want to Init and press shift+page and then gen on screen and just leave all values at 0 and commit?

shsCT, 2020-09-27:
Hmm interesting! Thanks - Will have a go at that. Might still be useful to have a quick one-push way to clean out existing ranges variations/probabilities in the heat of a live performance though. But boy do I love this sequencer!

westlicht, 2020-10-14:
I don't see a good way to put the functionality into the `INIT` function. Instead I opted to add a `Init Layer` generator for that function. The problem with using other generators is that it's not necessarily setting the layer to the _default value_.

Will be part of the 0.1.37 release.

closed 2020-10-15 by westlicht