westlicht/performer #237: Feature request? Recording one loop of a pattern in curve mode

natedey, 2020-10-14:
Thought I posted this on here, but maybe forgot to save.
Let's say I'm using MI Shades DC output in the Performer CV In to manually draw a shape. How can I hit record, draw in the curve within a single loop of the sequencer, and stop recording before the sequence loops starts to overwrite the curve? Can I get the Performer to run the pattern a single time and then stop? (Don't tell me to buy a Planar! :)

One obvious workaround is to put the clock through a VCA where the gate was opened for a single loop of the pattern and then off for long enough to shut off the clock and/or save the pattern, but could there be some internal "loop once and record" ability in the curve track?

hotelbachelard, 2020-11-26:
Hi there,
I was able to do this by setting up a 4-step Curve track set at 1 bar clock division (each step is 4 beats/1 bar long), and have step 1 do a max flat curve, and steps 2/3/4/ with nothing. That is routed to control Record on/off.  So Performer can now record for 1 bar and play back for 3 bars before recording again. Or, just stop after the one bar - but having that 4-step patched to control Record you can definitely stop recording after 1 bar.

natedey, 2020-11-26:
hey Bachelard, this is rayultine from Muff! Thank you for coming up with a great solution to this problem. :)

hotelbachelard, 2020-11-26:
hey! nice to see you here :) i actually had a lot of fun figuring that out. it did make me realize that the Record function does apply to all the Curve tracks when it's activated. It didn't affect the 4-step track that was switching the Record on/off because I think the step resolution was maybe too low/big (1 bar per step), but when I created more Curve tracks in the project, those also captured the incoming CV. So, one of my related requests to this one would be to be able to target the Record status on a per-track basis. OR, be able to assign CV and CV/Gate input source per track rather than per project.