westlicht/performer #239: Feature request: Drum Track

dComposer, 2020-10-14:
When programming drums I end up using a bunch of precious tracks just for their gate outputs. What I propose is a "Drum Track" which still uses the gate output but also turns the cv output into something that can independently shoot out it's own 5V gate. It could act like a normal note track with F1 sending you into the GATE1 page and F4 sending you into the GATE2 page (replacing the notes page). It's minor but it would let me consolidate drum tracks and use the outputs more efficiently.

westlicht, 2020-10-14:
This is not easily doable unfortunately. I would suggest you make your drums more interesting by using the CV out for some modulation :D

natedey, 2020-10-15:
Couldn't you just use a curve track with simplistic settings for this???

closed 2020-10-22 by westlicht