westlicht/performer #240: links in PDF manual

jasonnolan, 2020-10-16:
Hi Everyone;

I made a version of the manual that has hot links in it... because I need it if I'm ever going to keep on top of things. I figured perhaps someone else would want it too, so here it is.
[User Manual | PER FORMER eurorack sequencer.pdf](https://github.com/westlicht/performer/files/5391930/User.Manual.PER.FORMER.eurorack.sequencer.pdf)

ruidobello, 2020-10-16:
Great resource!!!
Thank you!!!
paz
jorge

westlicht, 2020-10-16:
Yeah, the missing links were a limitation of `markdown-pdf` that I used when first generating the PDF. It was annoying but there was no other way. In the meantime I found a better converter `mdpdf` which properly translates links. I spent some time to also improve formatting. I think the new PDF is pretty good now. Get it here: https://westlicht.github.io/performer/manual/manual.pdf

closed 2020-10-16 by westlicht

jasonnolan, 2020-10-16:
Yours is better than mine! Thanks. Big help. Cheers.