westlicht/performer #241: Launchpad request

deepiain, 2020-10-20:
Would it be possible to use button 6 for start/stop, instead of using a combination press of 8&7? 
Thanks
Iain

westlicht, 2020-10-22:
I prefer the shift combo because play/stop is not something I want to hit by accident.

deepiain, 2020-10-22:
The play / stop buttons on all my other kit are not behind a shift key.. can’t think of any time I’ve hit it by mistake?

westlicht, 2020-10-22:
If it's well separated then sure. But currently on the launchpad the two keys next to the 6 button are for fills and choosing the sequence mode. So unless I change the layout more it's still a bit of a risk, specifically with fill mode that is a performance feature.

GeoffMaciolek, 2020-10-28:
I suppose a much more complex "fix " for this would be to have a config file on the SD card to allow users to set button layout? (It would prevent the need for any UI changes). I imagine the default behavior would be "is there a config file? No? OK, write out one with the defaults."

I haven't looked at the relevant code (and my C++ skills are **very** limited), but I may take a peek!