westlicht/performer #242: Suggestion: Page numbers in manual

rolfjack, 2020-10-24:
Navigating in the manual can be challenging because there are no page numbers. 
I manually added to every page and on the index and it was a huge help.

So it would be grat for page numbers to be added to the pdf version of the manual so I/we do not have to add them manually whenever there is an update.

Thank you for all your great work.

rolfjack, 2020-12-29:
Hi. I am enjoying the v39 firmware manual, but once more have to enter pagenumbers manually in the PDF version.

Any chance of adding page numbers to the manual?

westlicht, 2021-02-08:
Unfortunately with the current toolchain to convert markdown to PDF it's not possible to add page numbers.

closed 2021-02-08 by westlicht