westlicht/performer #243: Configurable slide/slew behavior in CV tracks

natedey, 2020-10-29:
From muffwiggler:
>>>Has anyone gotten slide working properly in a basic note track? I can switch it on for the various notes, and hear some kind of change in the sound, but nothing like what I'd expect or want. Is there some kind of setting i've missed?

>>My understanding is that its not a "slide" in the mono-legato sense of the TB303, where the envelope of the incoming note is not re-triggered, and the previous note is held.
Adding a slide to a note in Performer still results in a Gate being triggered. It's just that the pitch of the incoming note slides from the pitch of the previous note over an adjustable proportion of the gate length.

>Yes. According to the developer, you need to change the gate length of the notes being slid between to 100% to get the full effect. Having tried this, it's still kind of lacking, but does work. Perhaps this is something that can be developed further in time.

After a month of using this module, the slide/portamento behavior is the only thing I yearn for improvement. I know it's so generic and cheezy to say so, but it's way less "acid" than the Audio Damage Sequencer 1, Stepper Acid, or even the TB-3PO applet in the Ornament & Crime Logarhythm/Hemispheres alt firmware. 

Some control over time/curve of slide or a way to make it uhh bouncy(?) would be so magnificent.

CedreLo, 2020-12-08:
A new layer with slide time per step could be great

westlicht, 2021-02-07:
Improved slide behaviour will be released with 0.1.40

> A new layer with slide time per step could be great

Not going to add this. You can achieve this by using internal routing from another track if you really want to experiment with it.

commit 5ed3445b (2021-02-07): improved slide signal generation

fixes #243
closed 2021-02-07 by westlicht