westlicht/performer #245: sending midi out of multiple tracks at once

whferris, 2020-11-01:
hello I am wondering if you can send and receive gate via midi from multiple tracks at once? so far all I can see is the gate source option in midi out page and it only allows me to choose one at a time. my hope is to send out my patterns into logic, edit them, and then send that midi info back to performer and record audio.

westlicht, 2021-02-07:
You can already do this today. You have 8 midi output slots available, so you can route all 8 tracks to different midi channels and record them externally to logic. Then you need to setup a new project that only consists of MIDI/CV tracks which essentially turns performer into a MIDI/CV interface. Or did I miss something?

closed 2021-02-07 by westlicht

whferris, 2021-02-08:
Hey thanks for response ! I was only able to assign a midi Chanel to the
whole module and couldn’t sort out how to assign to individual tracks but I
will try again . Very possible I missed something !