westlicht/performer #248: Buttons and LEDs stop working after boot

papastratman, 2020-11-08:
I've started having an issue with one of my Performers today. When booting, the LEDs that usually cycle across the buttons either don't cycle, or maybe cycle very slowly (possibly completing, or stopping part way). The boot process on the screen shows as completed, regardless of the button LEDs, but the buttons are non-responsive. The encoder works, the screen works, the clock in and reset in work. I was able to re-flash the firmware successfully. No buttons working, though. I've been using this module for three months, no issues until today. 
Has anyone else experienced this? Any suggestions?

westlicht, 2020-11-11:
Check the button led matrix circuit and components. Sounds a lot like a hardware issue. Probably one of the shift registers having a cold solder joint.

papastratman, 2020-11-11:
[image]
Looks like it was this guy here. I was thrown off since it's been working for months, but it was definitely an issue. 
Thank you very much for your help!

closed 2020-11-11 by papastratman