westlicht/performer #250: Additional scales

natedey, 2020-11-20:
Hey Simon,
Would users like to have additional choices for scales? The ability to add user scales is totally great, but there are comprehensive libraries of scales out there and it would be fun to have them hard coded in the Performer.
I have slowly been creating a doc where I am adding the necessary code in Scale.cpp for all of the scales listed in https://ornament-and-cri.me/predefined_scales/ 
Is this too many built-in scales and you prefer the elegance of only including the most widely used ones? Or would you like me to complete this doc and send it when finished?

natedey, 2020-12-14:
If anyone wants to double check my work or make sure display formatting and variable naming is correct, here it is!

NOTE_SCALE(ionianScale, "Ionian", true, 0, 256, 512, 640, 896, 1152, 1408)
NOTE_SCALE(dorianScale, "Dorian", true, 0, 256, 384, 640, 896, 1152, 1280)
NOTE_SCALE(phrygianScale, “Phrygian", true, 0, 128, 384, 640, 896, 1024, 1280)
NOTE_SCALE(lydianScale, "Lydian", true, 0, 256, 512, 768, 896, 1152, 1408)
NOTE_SCALE(mixolydianScale, "Mixolydian", true, 0, 256, 512, 640, 896, 1152, 1280)
NOTE_SCALE(aeolianScale, "Aeolian", true, 0, 256, 384, 640, 896, 1024, 1280)
NOTE_SCALE(locrianScale, "Locrian", true, 0, 128, 384, 640, 768, 1024, 1280)
NOTE_SCALE(pythagoreanScale, "Pythagorean", false, 0, 115, 261, 376, 522, 637, 783, 899, 1014, 1160, 1275, 1421)
NOTE_SCALE(eb-4Scale, "EB/4", false, 0, 128, 256, 384, 448, 640, 768, 896, 1024, 1152, 1280, 1344)
NOTE_SCALE(e-4Scale, "E/4", false, 0, 128, 256, 384, 448, 640, 768, 896, 1024, 1152, 1280, 1408)
NOTE_SCALE(ea-4Scale, "EA/4", false,  0, 128, 256, 384, 448, 640, 768, 896, 1024, 1088, 1280, 1408)
NOTE_SCALE(bhairavScale, “Bhairav", false,  0, 115, 494, 637, 899, 1014, 1393)
NOTE_SCALE(marwaScale, “Marwa", false,  0, 143, 494, 755, 1132, 1393)
NOTE_SCALE(yamanScale, “Yaman", false,  0, 261, 522, 783, 899, 1160, 1421)
NOTE_SCALE(kafiScale, “Kafi", false,  0, 233, 376, 637, 899, 1132, 1275)
NOTE_SCALE(bhimpalasreeScale, “Bhimpalasree", false,  0, 261, 404, 637, 899, 1160, 1303)
NOTE_SCALE(bairagiScale, “Bairagi", false,  0, 115, 637, 899, 1275)
NOTE_SCALE(tartvallottiScale, “TartVallotti", false,  0, 120, 251, 381, 502, 643, 758, 893, 1019, 1144, 1280, 1395)
NOTE_SCALE(13of22tetgen5Scale, “13of22tETgen5", false,  0, 140, 279, 419, 489, 628, 768, 838, 977, 1117, 1187, 1327, 1466)
NOTE_SCALE(mandelbaumScale, “Mandelbaum", false, 0, 162, 243, 404, 485, 647, 728, 889, 970, 1132, 1213, 1374, 1455)
NOTE_SCALE(magicin145tetScale, “Magicin145tET", false, 0, 191, 265, 339, 413, 487, 561, 752, 826, 900, 975, 1049, 1239, 1314, 1388, 1462)
NOTE_SCALE(quartaminor3rdsScale, “Quartaminor3rds", false, 0, 99, 199, 298, 398, 497, 597, 696, 796, 895, 995, 1094, 1193, 1293, 1392, 1492)
NOTE_SCALE(armoduesemieqScale, “Armoduesemi-eq", false, 0, 99, 198, 297, 396, 495, 595, 694, 793, 892, 991, 1090, 1189, 1239, 1338, 1437)
NOTE_SCALE(hirajoshiScale, “Hirajoshi", false, 0, 237, 431, 874, 1011)
NOTE_SCALE(scotbagpipesScale, “Scotbagpipes", false, 0, 252, 436, 634, 900, 1092, 1292)
NOTE_SCALE(thairanatScale, “Thairanat", false, 0, 206, 443, 673, 878, 1103, 1317)
NOTE_SCALE(sevish12on31edoScale, “Sevish12on31EDO", false, 0, 149, 297, 396, 545, 644, 793, 941, 1041, 1189, 1288, 1437)
NOTE_SCALE(11tetmachine6Scale, “11tetMachine6", false, 0, 279, 559, 698, 977, 1257)
NOTE_SCALE(13tetfather8Scale, “13tetFather8", false, 0, 236, 473, 591, 827, 1063, 1182, 1418)
NOTE_SCALE(15tetblackwd10Scale, “15tetBlackwd10", false, 0, 205, 307, 512, 614, 819, 922, 1126, 1229, 1434)
NOTE_SCALE(16tetmavila7Scale, “16tetMavila7", false, 0, 192, 384, 672, 864, 1056, 1248)
NOTE_SCALE(16tetmavila9Scale, “16tetMavila9", false, 0, 96, 288, 480, 672, 768, 960, 1152, 1344)
NOTE_SCALE(17tetsuprpyth12Scale, “17tetSuprpyth12", false, 0, 90, 181, 361, 452, 632, 723, 813, 994, 1084, 1265, 1355)
NOTE_SCALE(22tetorwell9Scale, “22tetOrwell9", false, 0, 140, 349, 489, 698, 838, 1047, 1187, 1396)
NOTE_SCALE(22tetpajarasy10Scale, “22tetPajaraSy10", false, 0, 140, 279, 489, 628, 768, 908, 1047, 1257, 1396)
NOTE_SCALE(22tetpajara510Scale, “22tetPajara5-10", false, 0, 140, 279, 489, 628, 768, 908, 1117, 1257, 1396)
NOTE_SCALE(22tetporcupine7Scale, “22tetPorcupine7", false, 0, 209, 419, 628, 908, 1117, 1327)
NOTE_SCALE(26tetflattone12Scale, “26tetFlattone12", false, 0, 59, 236, 295, 473, 650, 709, 886, 945, 1122, 1182, 1359)
NOTE_SCALE(26tetlemba10Scale, “26tetLemba10", false,  0, 177, 295, 473, 591, 768, 945, 1063, 1241, 1359)
NOTE_SCALE(46tetsensi11Scale, “46tetSensi11", false, 0, 167, 334, 501, 568, 735, 902, 1069, 1135, 1302, 1469)
NOTE_SCALE(53tetorwell9Scale, “53tetOrwell9", false, 0, 145, 348, 493, 696, 840, 1043, 1188, 1391)
NOTE_SCALE(12of72tetrodgrsScale, “12of72tetRodgrs", false, 0, 256, 341, 491, 555, 640, 704, 896, 1131, 1237, 1344, 1387)
NOTE_SCALE(trivalentzeus7Scale, “TrivalentZeus7", false, 0, 202, 496, 698, 900, 1195, 1396)
NOTE_SCALE(202tetoctoneScale, “202tetOctone", false, 0, 152, 449, 494, 791, 897, 1194, 1239)
NOTE_SCALE(313tetelfmadag9Scale, “313tetElfMadag9", false, 0, 260, 319, 579, 638, 898, 957, 1217, 1276)
NOTE_SCALE(marvelwooglummaScale, “MarvelWooGlumma", false, 0, 63, 298, 406, 491, 703, 789, 897, 1131, 1194, 1239, 1492)
NOTE_SCALE(topparapyth12Scale, “TOP-Parapyth12", false, 0, 75, 265, 340, 530, 605, 710, 901, 975, 1166, 1240, 1431)
NOTE_SCALE(16-ED2or3Scale, “16-ED_2_or_3", false, 0, 96, 192, 288, 384, 480, 576, 672, 768, 864, 960, 1056, 1152, 1248, 1344, 1440)
NOTE_SCALE(15-ED2or3Scale, “15-ED_2_or_3", false, 0, 102, 205, 307, 410, 512, 614, 717, 819, 922, 1024, 1126, 1229, 1331, 1434)
NOTE_SCALE(14-ED2or3Scale, “14-ED_2_or_3", false, 0, 110, 219, 329, 439, 549, 658, 768, 878, 987, 1097, 1207, 1317, 1426)
NOTE_SCALE(13-ED2or3Scale, “13-ED_2_or_3", false, 0, 118, 236, 354, 473, 591, 709, 827, 945, 1063, 1182, 1300, 1418)
NOTE_SCALE(11-ED2or3Scale, “11-ED_2_or_3", false, 0, 140, 279, 419, 559, 698, 838, 977, 1117, 1257, 1396)
NOTE_SCALE(10-ED2or3Scale, “10-ED_2_or_3", false, 0, 154, 307, 461, 614, 768, 922, 1075, 1229, 1382)
[... cut ...]
NOTE_SCALE(8-HD2Scale, “8-HD2", false,  0, 261, 494, 706, 899, 1076, 1240, 1393)
NOTE_SCALE(7-HD2Scale, “7-HD2", false,  0, 296, 557, 790, 1002, 1194, 1372)
[... 82 lines cut ...]
&12-HD2Scale,
&11-HD2Scale,
&10-HD2Scale,
&9-HD2Scale,
&8-HD2Scale,
&7-HD2Scale,
&6-HD2Scale,
&5-HD2Scale,
&4-HD2Scale,
&16-SD2Scale,
&15-SD2Scale,
&14-SD2Scale,
&13-SD2Scale,
&12-SD2Scale,
&11-SD2Scale,
&10-SD2Scale,
&9-SD2Scale,
&8-SD2Scale,
&7-SD2Scale,
&6-SD2Scale,
&5-SD2Scale,
&4-SD2Scale,
&bohlenpierceequalScale,
&bohlenpiercejScale,
&bohlenpiercelScale,

natedey, 2021-03-28:
Are saved user scales compiled into one file? Or saved as separate files? It would be awesome if the latter because then users could share libraries of custom scales and microtonal scales? If the former, I could add all of the above scales to the user scales file, but then wouldn't be able to share with others without overwriting whatever scales they have constructed on their SD card. 

Bonus points if the performer could eventually import/export Scala formatted files!

Also found a missing but obvious feature in the user scales menu that holding shift while turning the encoder increments the V of a tuning increment by +/- 0.1 V instead of the default 0.01 V

MyTVC-15, 2024-08-04:
Hello Natedey, thank you for your post. I have the requirement to quickly access church mode scales, melodic minor, harmonic minor. Have you managed to implement your scales on the Westlicht Performer? Cheers, Christof