westlicht/performer #253: Per-step gate offset

hotelbachelard, 2020-11-26:
Hi, 

I'd love to be able to do a negative gate offset on a step. Currently you can only go between 0 and 87. 5%, but you can't make the step trigger earlier. The screen also puts the 0% mark in the middle, so I assumed that you can go to negative values. 

Many thanks!

vafu, 2020-12-01:
Hi, I have a PR with negative offsets implemented and I need some help testing it, would you mind trying my changes out?
https://github.com/westlicht/performer/pull/234

hotelbachelard, 2020-12-01:
Hey there,
Sure! Thanks for asking

---------------Music: http://www.bachelard.bandcamp.com

hotelbachelard, 2020-12-06:
Hey @vafu - your latest version has a glitch. In the Step mode/page, normally, you can only turn on/off a step gate when you're on the Gate menu/page (1st tab, 1st item), and you need to press on the step + encoder when you're on any other  menu (Retrig, Note, etc). But with your latest version, you can activate gate on/off from any page. Not sure if that's what you want - I personally prefer to be able to alter the gate sequence only in the Gate menu, so it's harder to accidentally alter the gate pattern while I'm trying to just change note values or gate lengths, etc.

I reloaded the official 0.1.39 firmware to confirm, because that's what I remember, and confirm that's the case.

vafu, 2020-12-06:
@hotelbachelard yes, thanks! I've added this as my own preference but I accidentally added these changes to the firmware update, sorry about that, I'm going to remove that asap!

hotelbachelard, 2021-02-28:
Hey @vafu - I found a lil bug in your 1.39: the Gate Probability bar doesn't show up in your version. :(
[image]
it's there in 1.40 (and the official 1.39)
[image]