westlicht/performer #254: Mod wheel not working

simont-org, 2020-12-01:
Hi, I tried with many controllers but modulation wheel has no effect. Is it me, or it is not managed?
Maybe some settings not clear in the manual?
Thanks

westlicht, 2020-12-16:
Mod wheel (controller 1) is working fine for me. But if your controller sends NRPN it will not be detected by the MIDI learn feature. Is the issue here that MIDI learn is not working?

simont-org, 2020-12-16:
I've just checked my keyboard. Pitch wheel sends PITCHBEND with 16 bit value, while mod wheel sends CC 1 with 0-127 value.
But in performer nothing happens. Is there something I need to configure? Some mapping or flag, maybe? I am just using a midi/cv track.
To compare, I tested the same keyboard with the same connection with a Disting Mk4 in MIdi2CV mode, and mod wheel is perfectly read. Thanks

westlicht, 2020-12-17:
On a MIDI/CV track the mod wheel is not doing anything. Only pitch bend and aftertouch are supported (and output as CV). What should the mod wheel do?

simont-org, 2020-12-17:
For example it could be routed somewhere (gate, slide, etc.). Or just transpose in a fixed way like other environments. I mean, pitch wheel changes pitch and come back. Mod is not spring loaded so you can move it and transpose.
When you say that on a MIDI/CV track is not doing anything you mean that it does on a note track?

westlicht, 2021-02-09:
You can already setup routings for _SlideTime_ and _Transpose_ on the MIDI/CV track. So you can take the mod wheel and route it to one of those parameters. Support for gate length might be a nice addition.

simont-org, 2021-02-09:
Yes, it works. Only thing I noticed is that the transpose is "discrete", you can hear the discrete change between notes while you move the mod wheel. But I guess I have to work on slide settings