westlicht/performer #260: Next Pattern Fill issue

hotelbachelard, 2020-12-05:
Last night I was delving a bit more into the Fill modes, and discovered that if you chooose Next Pattern as your Fill mode, the "next" pattern that is played when you activate Fill conforms to the clock division of the first/original pattern.

E.g. Pattern 1 is set at 1/8 clock div, Pattern 2 is set at 1/16 div. When Fill is activated for Pattern 1 in Performance Mode, Pattern 2 is played, but at 1/8, not 1/16.

Anyway to fix that?

napoleonthegreat94, 2020-12-25:
This would be a realy nice feature, i wonder if it is possible

westlicht, 2021-02-09:
Unfortunately it's not something I can do with the current memory restrictions. As mentioned before, the snapshot system can be used to easily set a savepoint to easily jump back to.

closed 2021-02-09 by westlicht

napoleonthegreat94, 2021-02-09:
Fair enough, thanks for looking into it nonetheless