westlicht/performer #269: Feature Request: Overdub/overwrite setting as a route

shsCT, 2020-12-22:
Hello! Would it be possible to add the recording type (overdub/overwrite) to the midi routing menu? Both are so useful in live performance for different reasons - it’d be great to assign a midi controller to toggle between them while playing live. Thanks for the consideration!

westlicht, 2021-02-09:
Will look into it, just added _Play Toggle_ and _Record Toggle_ routing targets with the 0.1.40 release.

shsCT, 2021-05-24:
That's a great new addition. 

Still hoping though that this route can be added; it's really amazing to be able to record in both manners especially when building up a track live. It's a little clumsy to have to jump into the project menu and out again at speed when you're performing live; I go back and forth when say adding to or replacing a bass line - and both are incredibly useful for different results! This would make this sequencer uber-powerful in live situations!

Hoping hoping hoping :) :) :)