westlicht/performer #272: Record not receiving MIDI from TRS input

rolfjack, 2020-12-29:
When recording in any mode (step/overdub/overwrite) I am unable to get the Performer to respond to midi events coming in on the TRS input. I have experimented with channel settings, omni etc. Makes no difference.

I am getting midi input on TRS in general. Midi/CV tracks are working, routing is working, learn in routing is working etc. Arpeggiator is working through TRS midi.

**But when trying to record to a track using the TRS midi input: Nothing.**
When attaching a keyboard via USB it works without a hitch.

As I recall, this used to work, so perhaps a bug has been introduced with the  updates made to MIDI in the last versions?
Using USB means I cannot simultanously use my Launchpad as well as other drawbacks, so having midi through USB for recording is a big advantages.

Kind regards, Rolf

westlicht, 2021-02-08:
Recording via TRS MIDI works fine for me. Can you try again with an empty project?

hotelbachelard, 2024-02-07:
Hi, I just encountered same issue on a project I'm working on, but discovered that i can't record incoming MIDI from the TRS input if I have any of the tracks set as a MIDI/CV track. Even if I want to capture the MIDI on another track it won't let me. In my case, I have Tracks 6-8 receiving MIDI from CH 1-3 of a Yamaha QY300 sequencer, but I also wanted to capture parts of the pattern coming from MIDI ch 1 onto, say, Track 1 of the Westlicht. 

I suppose that makes sense (although i don't have full clarity of why it does...something about MIDI loop or feedback I suppose?), but also, it would be nice to be able to use some tracks of the Westlicht as real-time MIDI/CV converter (which it does very well), and also still be able to capture incoming MIDI on another track.