westlicht/performer #275: Feature request: Note range variations: interval, chord, scale

hotelbachelard, 2020-12-30:
Was wondering if it's possible to constrain the Note range variations with options for **interval** (e.g., note range 3" chooses only between the original note and its minor 3rd, not all 3 semitones), chord, and/or scale. That would make the note variations a bit more musical.

Mauv3, 2021-01-08:
If you select a scale, note range will pick notes from that. If you want a more musical result you can create a custom scale containing only certain notes, for example root, 3rd, 4th, 5th, leaving out the dissonances like 2nd and 7th.

hotelbachelard, 2021-01-09:
> If you select a scale, note range will pick notes from that. If you want a more musical result you can create a custom scale containing only certain notes, for example root, 3rd, 4th, 5th, leaving out the dissonances like 2nd and 7th.

Good tip with the user scale, thanks! for

westlicht, 2021-02-08:
The idea always was to use this feature together with scales. There are two many additional parameters that would be needed to get more control of that.

closed 2021-02-08 by westlicht