westlicht/performer #277: Browser simulator / test drive: alt key not functional on OSX browsers

infovore, 2021-01-12:
On the 'testdrive' version of the simulator, the alt key is used to simulate the "page" key. However, on all Mac OSX browsers I've tried, this key does not register.

This feels like it's possibly an `emscripten` issue rather than an SDL one - the SDL keycode in the simulator C++ is correct. I note a recent emscripten update may have fixed this, but haven't quite worked out how emscripten is included into the project.

I appreciate this is a minor issue compared to issues with the sequencer firmware itself.

westlicht, 2021-02-06:
On Macs you need to use the "Option" key. Works fine here on Chrome, Firefox and Safari.

westlicht, 2021-02-06:
Not sure what the issue is you run into but maybe try again or give me more instructions on how to reproduce it. Closing this for now.

closed 2021-02-06 by westlicht

infovore, 2021-02-07:
I should have been clearer: I _was_ using the option key (which is also marked 'alt' on macs, hence not clarifying it). In both latest Safari and Chrome: nope, no dice.

Never mind. It's just the simulator.

westlicht, 2021-02-07:
I'm still running Mojave, are you on Big Sur?