westlicht/performer #281: Add A Repository For The Manual

BerberBerthold, 2021-01-27:
There are many undocumented features that are missing from the documentation.  It seems that Simon struggles to get himself to do this, so it might be a good idea if the manual were in a git repository on github, so that the community can add information etc.

westlicht, 2021-02-06:
Can you please list the things that are not in the manual? I'm actually very much trying to document every feature I add or change.

And of course the documentation is in a repo, you can find it here: https://github.com/westlicht/performer/tree/gh-pages/_pages/manual

BerberBerthold, 2021-02-06:
I don't know why I was too dumb to find the manual in this repo.  I did try to find it, though.  Sorry.

Things that are missing are off the top of my head are:
- Euclidean Sequencing
- More specific info on the Launchpad.  As in stuff like how the data is represented on the pads.
- Info on the size (HP) of the module and maybe some other specs
- A photo of the module, or just what the module looks like at the very top of the manual including all buttons and I/O (this also applies to the index page of the website)
- I think the way ratcheting (retrigger) works can be explained a bit more precisely maybe.  Or I'm confused.  What I mean is that note length needs to be long enough for a 4 trigs per step, and that if the length is set to full, then the last of the 4 trigs will not legato over to the next step.
- One of the most watched videos on the Performer talks about how you can use the module as a Pitch quantizer.  Even though this is not really a bare feature but more of an example of a use case, examples of use cases are a good idea for manuals.  It could also be phrased as a feature.  I mean, when reading the manual as is, it should be quite immediate to realize that setting CV in to transpose notes between discrete points values within a scale is basically directly pitch quantization for the chosen scale, it stil requires a thorough reading to get this.  Actually, I have a question:  Is this a real "quantizer"?  In a major scale, +7 would correspond to one octave higher, but in the chromatic scale +7 doesn't correspond to an octave higher, but they would correspond to the same input CV, correct?  Or does the Performer measure the VC in voltage and quantizes it to the closest voltage in the defined scale table (in which case this would be a true quantizer)?  I should try this out myself haha.  But yeah stuff like that could be documented.

westlicht, 2021-02-06:
> Euclidean Sequencing

Generator is not documented as I still consider the feature "incomplete" and work-in-progress. But at this point in time it probably isn't anymore so I should add documentation I think.

> More specific info on the Launchpad. As in stuff like how the data is represented on the pads.

Ok, that part might be lacking a bit yes.

> Info on the size (HP) of the module and maybe some other specs

There is a general features section, but yes, HP is missing there. Full specs are on the main page https://westlicht.github.io/performer/ but it would probably make sense to have them in the manual as well.

> A photo of the module, or just what the module looks like at the very top of the manual including all buttons and I/O (this also applies to the index page of the website)

There is a diagram of all buttons where the UI concepts are introduced. But a beauty shot of the module in the beginning of the manual would probably make sense :)

> I think the way ratcheting (retrigger) works can be explained a bit more precisely maybe. Or I'm confused. What I mean is that note length needs to be long enough for a 4 trigs per step, and that if the length is set to full, then the last of the 4 trigs will not legato over to the next step.

Yep seems like that's missing.

Thanks for the feedback. This is very welcome. I know the manual is quite "technical" in its current form. I kept it like that to make it easier to keep up-to-date with the firmware changes. This gets harder when we talk about workflows and applications, as things might be missed when updating the firmware. I always wanted to do tutorial videos to show off the workflows, which would be a much better resource. I just haven't gotten to it really. Also some other people have started doing tutorials on youtube which is really welcome!

westlicht, 2021-02-06:
Also, feel free to open PRs to the `gh-pages` branch fixing those issues.

BerberBerthold, 2021-02-06:
i know that there is a diagram of all the buttons jn the manual, but it doesn't include the inputs and outputs.

i will make some PRs when i have time to get to it.

also, i don't have the module with me right now, i eill tested once i can play around with it again, but how does the CV-in get translated to "+1" ,"+2", "+3" etc for "transpose"?  that should also be documented in my opinion, as mentioned in my above comment.

thank you for keeping this project alive!

westlicht, 2021-02-06:
> also, i don't have the module with me right now, i eill tested once i can play around with it again, but how does the CV-in get translated to "+1" ,"+2", "+3" etc for "transpose"? that should also be documented in my opinion, as mentioned in my above comment.

this all depends on how you setup the route. you can configure both an input range (from CV) as well as an output range. so you are free on how you map the CV to the actual parameter. for transpose you for example can have -1V..1V mapped to -12..12 semitones to get V/oct. but in general routings are agnostic to any scale settings, they simple map an input value to a parameter.

tl072, 2021-02-12:
This is a really good idea. I am sure there are clarifications that I could have contributed. I will see if I can remember things that took some figuring out. One thing that I still do not fully grasp (and have not had time to experiment with) is the midi routing with a second track providing the velocity. As an aside I am going to make a feature request that velocity can be set on the track providing the pitch but that is a different story. :)