westlicht/performer #285: UPDATE.DAT issues

ronnyjamesdisco, 2021-02-10:
I had issues with the last three updates 1.38-1.40  using the UPDATE.DAT file copied to SD-card. 

While the bootloader (power-on /w encoder pressed) was updating suddenly the screen went blank leaving the device essentially bricked. Reflashing bootloader and sequencer hex files via ST-Link/JTAG got it back alive (and in general it is working without issues). Any clue where to look for the reason of this problem?

orys, 2023-02-16:
I'm in the same boat right now and can't update to 1.42 from 1.41. 
I have no programmer either. :(

ronnyjamesdisco, 2023-02-19:
@orys if you are referring to the issue of your other post https://github.com/westlicht/performer/issues/376#issue-1588207068 
then I don't think we are in the same boat

closed 2025-09-04 by ronnyjamesdisco