westlicht/performer #289: CV/Gate track with scale set to "Voltage"

natedey, 2021-02-26:
Hi Simon,
I thought the "Voltage" scale might be a handy way to dial in quantized voltages for an arbitrary tuning. However, the note settings seem to only have a record/dial resolution to 0.1V. That effectively makes it a 10-TET scale, right?
Would it be possible to allow the encoder to set voltages in this scale to two or three digits? Or would that create a situation where the encoder would need to be turned for an uncomfortably long time to dial in precise values? 
Thanks!

hotelbachelard, 2021-02-28:
Good one. I'd be interested in this tweak!

westlicht, 2021-03-08:
So there is an internal resolution of 7bits (0..127) for note values. Those are mapped to voltages by the selected scale. The _Voltage_ scale has 0.1V resolution because the intention was to dial in rough voltage values only. You should use one of the user scales for setting up custom scales. Those can have arbitrary voltages down to mV resolution and also allow you to define what an _octave_ is so you only specify the notes through one octave which then automatically gets repeated up and down.

westlicht, 2021-03-25:
Closing this as there is already a very powerful facility to make scales with arbitrary tunings.

closed 2021-03-25 by westlicht