westlicht/performer #290: MIDI note entry not filtered?

tokkeherfst, 2021-02-28:
When editing notes with midi in connected, it seems that any midi note events coming in are changing the held step note value, regardless of the "MIDI record" setting on the Project page; neither Off or restricting to a certain MIDI channel helps. 
"Midi Record" does seem to have the correct effect on recording though. So I suspect the filtering code is not applied on note entry.

This is even the case with a completely INITed project. I am running 0.1.40 compiled fresh from github.

westlicht, 2021-03-08:
Yep, this is by design. The idea is that you can connect a MIDI keyboard and use it for note entry by holding step keys and pressing keys on the MIDI keyboard. This is completely decoupled from recording.

tokkeherfst, 2021-03-09:
Ok, I was trying to use one of the note tracks on the Beatstep Pro as midi keyboard, while also getting clock from it over MIDI. The problem is when the sequencer is running, notes on any track will impact step note edit on the Performer... also the drum track. I can get by with using CV but it's not ideal. So I wonder if you might consider separate MIDI filtering for the note entry?

BTW Love the performer, especially in combination with a launchpad! In terms of open source eurorack, it is a fantastic contribution. Thank you for sharing this with the world!

geoffreypb, 2021-04-12:
I am experiencing the same workflow trouble currently. I assumed it was a bug with MIDI monitoring but I see now it is by design - which makes sense in your described use case. Additional options for MIDI filtering would be greatly appreciated. 
Thanks for all of your hard work!

momoche, 2021-05-17:
would be great to have the option yes. i like the design of recording or even adding to existing sequences live, but also inconvenient in a lot of cases. my workaround is to leave an empty channel but that is a shame.

dblU2020, 2022-01-06:
Does anybody know a workaround for this?
Could this be filtered software-wise?
Would like to use my MIDI keyboard and another sequencer (DAW) for several synths and the Performer and synths being attached to it. Cheers

franq87, 2022-02-04:
I'd be grateful is a midi filtering solution could be implemented :)

Jeanbender, 2022-03-11:
Hi ! I have the same issue, and i'm glad to know it's not a bug. But sadly when not used with a keyboard, it makes less usefull the westlicht midi abilities.. Mine is slaved by an octatrack. But i can't make my own sequenced tracks, as all notes are based on the octatrack one.. That would be great if an option about that would be purposed ! thanks !

SonicMonk, 2022-04-06:
The only way to restore real-time recording for what you are playing on your midi keyboard seems to be to go back to the previous firmware version.

dblU2020, 2022-06-12:
Which firmware version would that be?