westlicht/performer #296: iterate runMode when pressing `Page` + `Mode`

modularev, 2021-03-20:
Current behaviour:
Pressing `Page` + `Mode` jumps from `FORWARD` directly to `RANDOM WALK`

Desired behaviour:
Pressing `Page` + `Mode` iterating through all `RunModes` available

I wasn't able to figure out where to look in the code, if you could point me in the right direction I'd happily test it and make a PR

modularev, 2021-03-20:
I just noticed that not only the ´RunModes´ are affected, but all sequence settings

westlicht, 2021-03-25:
I think you don't understand the UI here. When you press `PAGE` + `MODE` you see a bunch of LEDs light up (S1-S6) with the red one being the currently selected mode. Press those buttons to select the new mode.

But it seems there is also a hidden bug because pressing `PAGE` + `MODE` and then `MODE` again should actually do nothing, because it's not one of the assigned buttons to select modes.

commit da9cecd2 (2021-03-25): fix quick edit

fixes #296
closed 2021-03-25 by westlicht