westlicht/performer #301: Feature request: 1-2 more bits of resolution for the offset layer

tl072, 2021-03-26:
As I said in my other feature request today, I think I have read that bits for steps are valuable things and great care has to be used in allocating them. 

With that said, I find that the 12.5% increments a little more coarse grained than I would prefer. Usually playing with swing I can get the groove I am looking for but if there is 1 to 2 more bits that could be spared to the offset layer to get a bit more fine control I think it could be a very nice addition.

Just to add, I really am enjoying this sequencer. Please consider this an icing on the cake feature request!

Chai-Baba, 2021-04-07:
Thats also a huge and profound thing i would love to see too. I don`t know how often i have read parts of the manual in the expectation to find the usefull hint, how to dial in decimal values. For me it makes things really hard and often unusable when i like to set the note length and gate delays right to my taste, so that it gels to the swinging feel of the rest of my modular setup. If it is too coarse it remains often an unusable feature for me.

jaffasplaffa, 2021-08-03:
I would have to agree here, 12.5% steps are a bit rough for the delays, some finer adjustment would be great.

I was thinking the other day, that it would be lovely if we could for example hold down the encoder while setting a value, to get finer resolution. 

That could be implemented for all the step parameters :)