westlicht/performer #304: OLED screen hum/noise issue

jackpf, 2021-04-13:
So, the OLED screen seems to cause a bit of hum on the outputs of the Performer, and actually seems to feed back into the power supply and bleed into other modules on the same supply.

The noise isn't audible at normal levels, but when amplified it's quite noticeable. It also seems related to the number of pixels that are lit, eg. switching pages/enabling steps causes different levels of noise.

Switching to a linear power supply, and making some tweaks to the UI to reduce the number of pixels used has made a pretty good reduction to the noise, but hasn't got rid of it completely.

I've found some other bits of information about these kind of OLEDs causing the same issue, eg:

https://forum.mutable-instruments.net/t/digital-noise-from-oled-now-with-examples-read/3306
http://ixox.fr/forum/index.php?topic=69470.0
http://ixox.fr/forum/index.php?topic=69281.0

Just thought i'd share this here, in the hope that anyone might have any more ideas on how to reduce this noise.

Cheers!

tl072, 2021-04-15:
Not time to read the forum posts you have sent (will come back to later) but this might also be relevant: https://www.muffwiggler.com/forum/viewtopic.php?f=17&t=233927

The main take-away is maybe linear power supply but there are some other interesting things in there. No simple fixes as far as I can recall. There is a discussion of using massive grounding ground returns. It might be possible to graft something onto the westlicht pcb but that would be well beyond my skill-set.

jackpf, 2021-04-15:
Ah interesting read, thanks.

For sure the power quality plays a big role.

I have tested with a few power supplies to see how the noise is affected:

The 4MS row power with switching adapter was the worst.

Powering from my linear bench supply gets rid of the noise completely.

I then tried with a ZDROJ linear eurorack supply, and the noise is dropped by a few dBs compared to the 4MS, but not gone completely.

Now, I'm not sure if this is just because my big bench supply is cleaner than the ZDROJ.
I'm currently using the ZDROJ outside of my case (I need to get a new case for it to fit), so the wiring is quite long to the busboard so impedance is not great, and I also don't have a chassis ground connection. So i'll try as well with this and see if it gives any improvements.

I'm hoping someone might come along and just say it can all be solved by placing a resistor somewhere on the Performer, but that might be wishful thinking :D

Pipstapee, 2021-04-17:
A lot of hum issues are caused by bad grounding normally caused by double insulated PSUs like the use of wall wart and not the three pin plug type direct to your system. Seek professional advice on this though if you're not happy with electrics. Muffs has loads of thread open on this or look up Graham Hinton...

toddsines, 2021-08-02:
I also have a Row 40 power supply, chained off a Befaco PowSkiff, powered by a Lenovo 90W 20v laptop switchable PSU, and it's pretty insane how loud a white OLED emits noise in this system. :(

ftdBs, 2021-08-17:
I had this problem and where I live there isn't proper earth on the building, so gounded electrical cables would do nothing. 

Linear supplies didnt make much difference aswell. I guess they need to have a really high end regulation or rejection ratio or something. 

What worked for me was placing a separate power supply (again switch mode like the rest of mine) for my digital modules, decoupling analog and digital entirely and the noise is fully gone.

jackpf, 2022-02-16:
Just in case anyone's interested, I made some modifications to the firmware to reduce the pixels of some elements (menu items and sequence squares), and also added settings for brightness and a screensaver: https://github.com/jackpf/performer/pull/5

These help with the noise (but obviously doesn't get rid of it completely)

Pipstapee, 2022-02-19:
I've never encountered any noise issue running a pair of RS-75-12s in series but the brightness feature is really nice to have if you've used a blue Oled...
I'm not a fan of blue but needs must and i'm not patient enough to wait for stock.

dblU2020, 2022-02-25:
I am having the same issue. And unfortunately I have not found a solution.
Could changing the OLED Display make a difference?

da6353, 2024-02-13:
Per display datasheet: https://newhavendisplay.com/content/specs/NHD-3.12-25664UCB2.pdf, suggest to use 'Jumper option #2', which apparently bypasses the internal (and possibly noisy) boost converter, but would require some mods to then power the display.