westlicht/performer #307: Westlicht clock offset

aledotm, 2021-04-22:
Hello, I'm having a clock issue with my brand new Performer (not assembled by me but by a professional); 
I was testing the sequencer by simply sending a gate from the performer to a kick drum and syncing it up by ear with ableton's metronome at the same tempo (125bpm) since Live offers a way to sync up things manually, I've noticed that the performer's clock would go off grid after a few bars and slowly going offset after a bit. I repeted the same test with metronome from ornament and crime and the kick would stay perfectly on grid after minutes. I also repeted the test by sending clock from O_c to the performer and still the clock seems to be unstable...What could this problem be related to? 
Thanks a lot in advance.

jackpf, 2021-07-08:
I've also noticed the clock output can be quite unstable.
Some modules don't mind it, but some modules (particularly digital delays) really go crazy when fed a clock from the performer.

I'd guess it's not being smoothed out so well.

The gate outputs are super stable though, so sacrificing one of them to use as a clock usually works.

jackpf, 2021-07-08:
Actually scratch that, increasing the the output pulse time seems to have fixed it