westlicht/performer #318: Feature change request: Constant Pattern Sync

controller700, 2021-06-03:
When jaming with the Performer I find it quite odd that we have to hold SYNC[F2] + S[1-16] to change a pattern synced to the next beat. 

All sequencers I know work exactly the other way around and changing patterns to the next musical measure "1" is the standard mode (see Beatstep Pro, TR machines, Elektron, 303 etc.).

Would be nice if we could select a standard behavior SYNC/INSTANTLY on the pattern page. 
For example LATCH [F1], SYNC [F2], INSTANT [F3], SNAP [F4]. 

I´m sure this feature is very easy to implement and hope it can find it´s way to the next update soon.

boonh, 2021-06-05:
+1 for this.

madeinspace, 2021-06-22:
+1 please!

toddsines, 2021-09-16:
Yes Please!

controller700, 2021-09-19:
Yesterday we received a fine little update and lots of issue cases have been closed - thanks for that! 
I still have high hopes that one day some clever guy will implement this feature ´cause it´s still a mystery to me how this could be overlooked (taking into account that Mr. Westlich came from the Arturia BSP).

Lokua, 2022-03-18:
With all due respect and in defense of how it currently works, all other sequencers got this wrong, and I've been saying that for a loooong time. The argument that "all other X does it Y way" is a great argument when you know someone just totally messed up , but in this case it overlooks one of the most elegant performance mechanisms available to an electronic musician, which is:

Patterns should be able to be played like keys. This is electronic music, and playing patterns like keys is an evolutionary step forward for sequencers as real instruments. There is no way to do that comfortably while also holding shift. If you think about it, being able to play patterns like keys means things like FILL mode are redundant. Just use patterns for fills. Oh want to inject a transposition right now, for just a beat? Bam. Hit your transposed pattern and then on the next beat go back. Need a quick snare rush? CRACK. It would crazy uncomfortable to do that while holding SYNC while you're performing live, and after all, what is this device called?

That said, I can totally understand how this would be frustrating if that's not your style, and don't see any harm in having this feature, just as long as my precious and hard to find instant pattern changing functionality can remain.

controller700, 2022-03-18:
> With all due respect and in defense of how it currently works, all other sequencers got this wrong, and I've been saying that for a loooong time. The argument that "all other X does it Y way" is a great argument when you know someone just totally messed up, but in this case it overlooks one of the most elegant performance mechanisms available to an electronic musician, which is:
> 
> Patterns should be able to be played like keys. This is electronic music, and playing patterns like keys is an evolutionary step forward for sequencers as real instruments. There is no way to do that comfortably while also holding shift. If you think about it, being able to play patterns like keys means things like FILL mode are almost redundant (don't get me wrong I love FILL mode). Just use patterns for fills. Oh want to inject a transposition right now, for just a beat? Bam. Hit your transposed pattern and then on the next beat go back. Need a quick snare rush? CRACK. It would crazy uncomfortable to do that while holding SYNC while you're performing live, and after all, what is this device called?
> 
> Anyway, that's my opinion, but also one of things that makes this device great, but I can totally understand how this would be frustrating if that's not your style, and don't see any harm in having this feature, just as long as my precious and hard to find instant pattern changing functionality can remain.

Well, we´re talking about two different ways of working. I already made a suggestion how to implement both methods of pattern selection in the 1st post. 
"Would be nice if we could select the standard behaviour SYNC/INSTANTLY on the pattern page.
For example LATCH [F1], SYNC [F2], INSTANT [F3], SNAP [F4]."