westlicht/performer #322: Polyphonic MIDI

bithod, 2021-06-22:
Having the ability to play 3 or more notes per MIDI channel would be amazing. I am not sure how this would/could be implemented but using up 3 whole tracks for chords eats up nearly half of the respective outputs.

I'd just love to have a MIDI track that kicks polyphonic MIDI out of the MIDI output. :-)

ftdBs, 2021-08-17:
Agreed! Using different track channels for composing polyphony is perceptually mind bending. Same goes for cv tracks, Once I have 4 tracks doing 4 voice polyphony, it's just difficult to keep track of what is going on. Note names don't really make sense unless you know hardcore music theory (like able to sight read on the spot or something) + switching to another track just makes it even more difficult and eats up pattern space. If it was possible to handle all 8 cv events on one single page, configured as groups of note / chord / velocity etc events, we could simply use rest of the tracks for fancier things.

toddsines, 2021-09-20:
Also in favor of polyphonic MIDI. As I am currently using half my Performer's tracks for rhythmic sequences, I'd rather use a MIDI to Trigger converter to do arrangements for drum parts. How you would incorporate Gate or Repeat probability on to these tracks, that's another question. 

Not quiet your question, but related: my current method to get more outputs is to "cheat": -5V gate outputs for Kick and Open Hi-Hats on the Gate output for that track; +5V CV outputs [but no gate] on the same track for snare/clap and Closed Hi-Hats on the CV output on the corresponding track.

bobbyplubell, 2022-03-21:
This would be fantastic. Any ideas for how this could be handled in the user interface?