westlicht/performer #330: Launchpad X Not Working

BashoJax, 2021-08-25:
Launchpad X with newest firmware is not working on Performer 0.1.40.  Just got the performer and launchpad, so have not tested with earlier version, though I’ve seen comments from others saying they could not get it to function.

If there are any details or testing with the X needed I’m available.

closed 2021-08-25 by BashoJax

CT-030, 2021-09-25:
Hello,I was just wondering how did you solve the problem ?

BashoJax, 2021-09-25:
Hey there.

So, it’s really odd to get it to work.  I have the launchpad x plugged into
the USB port when the module powers on.  At this point, the launchpad x has
a random pattern of limit pads.  I think press the custom button on the top
row.  Then I press then Send B button on the right side.  Then I press the
Capture MIDI button.  At this point, it starts to behave as it should
interacting with the performer.  Holding down any of the arrow buttons on
the upper left change to a mode selection, etc.  the documentation at this
point lines up with the launchpad behavior for the most part.

I haven’t found any documentation about the steps to do this, just kind of
fumbled my way into it.

Hope that helps!

CT-030, 2021-09-26:
Okay, thanks for all the tips and your helps, I will try this.
Cheers :)