westlicht/performer #335: USB TX not sending clock

Lokua, 2021-09-19:
I don't think MIDI clock is being sent with USB TX like it does with MIDI TX. I have tested with both an Arturia Keystep as well as a Faderfox EC-4 which both receive clock fine from DAW using same cables. Is USB TX expected to send clock info? I am on the v0.1.41 firmware which gave me hope considering it addresses USB issues but no luck for me.

westlicht, 2021-09-20:
Did you enable USB TX in the clock settings? Instead of testing with other other devices, it's better to use a MIDI monitor application to see if the messages are actually being sent. I would be a bit surprised if clock wasn't sent.

westlicht, 2021-09-20:
And it definitely has nothing to do with power over usb.

Lokua, 2022-03-18:
Hello. Sorry for not getting back sooner. Had to have been user error. Thanks for the support :)

closed 2022-03-18 by Lokua