westlicht/performer #348: How to connect the iPad to the performer for Midi2CV?

raagulmanoharan, 2021-12-15:
I'd like to use the perfomer as a Midi2CV module, But im not able to make it work on my ipad via the USB cable using this Camera adapter. What Im i doing wrong, whats the right workflow for this? Please help
[image]

GeoffMaciolek, 2022-02-02:
I don't think the Westlicht Performer can be a USB slave device - only a USB Host? You';l want a midi/USB adapter, then plug that into the Performer midi ports; you'll need a DIN-to-TRS adapter too, probably. (I have **not** used the Performer for midi-to-cv, so I can't speak to that side of it!)

Lokua, 2022-03-18:
The Performer can indeed accept USB MIDI devices, however it is known to not work with some. The manual has a section about this: https://westlicht.github.io/performer/manual/#appendix-usb-midi-devices

Looking at this issue, however, it looks like a patch went out that addresses some of the issue:
https://github.com/westlicht/performer/issues/127

So perhaps libusbhost has other issues, IDK. Good to make sure you're on the latest version that contains that patch if you aren't.