westlicht/performer #351: Launchpad x

jharper603, 2022-01-27:
For some reason my launch pad x and the performer aren't automatically synced

mntbighker, 2023-01-07:
I ordered a Launchpad X specifically for the Perf|ormer. The gate sequencer grid is in a jumbled seemingly random order bunched up in the lower right corner of the grid. In general it does not respond to many of the button presses indicated in the Westlicht manual as expected. Perhaps Novation has altered the sysex crap? I updated the LP firmware on arrival. And I have the latest Perf firmware. I also own the LP Pro, and that grid seems to respond as expected, so this appears to be a specific issue with the "X" model.

mntbighker, 2023-01-09:
I spent much of the weekend getting the firmware to build. I made a LOT of uneducated Google guesses, but shockingly I got it to build. And with the change I propose above ( from 0x0d to 0x0c ) the LP X now starts up and mostly works as expected. The navigation key combos still don't work, but the rest seems to function like the LP Pro mk3. 

In order to get it to build I had resort to newer versions of OpenOCD and the ARM Toolchain. This broke the code in many places that I had guess about fixing. I'm sure I have forgotten half the crap I had to change to get the compile to complete. Everywhere the was an ld -Map function I commented the line because the ARM ld no longer allows you to link to an exe. There were a few places I had to change [&] to []. I never got "make test" to work either.

Start testing: Jan 08 18:57 PST
----------------------------------------------------------
1/8 Testing: TestSerialization
1/8 Test: TestSerialization
Command: "/home/vosadmin/performer/build/stm32/release/src/tests/unit/core/io/TestSerialization"
Directory: /home/vosadmin/performer/build/stm32/release/src/tests/unit/core/io
"TestSerialization" start time: Jan 08 18:57 PST
Output:
----------------------------------------------------------
/home/vosadmin/performer/build/stm32/release/src/tests/unit/core/io/TestSerialization: /home/vosadmin/performer/build/stm32/release/src/tests/unit/core/io/TestSerialization: cannot execute binary file

<end of output>
Test time =   0.00 sec

mntbighker, 2023-11-16:
It looks like maybe someone has updated the code recently, so bumping this hoping it gets fixed officially.

rawsound, 2024-02-01:
I have had the same problem and change Launchpad X to Legacy mode works to me:

"Legacy Mode may be accessed by entering the Setup Menu (hold Session button), then pressing the purple Scene Launch button on the right hand side of the device. Now press Session to enter Legacy Mode. All buttons and pads will be unlit and may be lit via MIDI messages (see detailed release notes for layout)."

rawsound, 2024-02-01:
Programmer mode works perfect.