westlicht/performer #352: Feature request: using keys as a mini keyboard to play live

franq87, 2022-02-04:
Could it be a possibility of implementing some option to use the keys over a track like a short of mini keys? something in an elektron way where you can play with the keys and record the sequence you are playing or just using as a live keyboard for the track selected, even only beeing 16 keys it would be sooo useful to me

jbjhjm, 2022-03-07:
If I understand you correctly, the live playing is already possible.
Set a note for each of the 16 steps, and then while in note mode, pressing a key will send the according CV. 
I think it is only possible if the sequencer is not running though.

franq87, 2022-03-07:
> If I understand you correctly, the live playing is already possible. Set a note for each of the 16 steps, and then while in note mode, pressing a key will send the according CV. I think it is only possible if the sequencer is not running though.

This is a clever solution, only thing is you cannot write a sequence there, but yeah it helps if you want to use a track as keyboard only