westlicht/performer #353: GEN tab behaviour

Mauv3, 2022-02-16:
Hi! I was wondering if it would be possible for the per|former to pick a random pattern by itself instead of choosing manually one form the hundreds available.
Maybe with a press of a button the per|former spits out a pattern and then the smoothness and scale can be adjusted manually as usual.
Thank you for your time, hard work in making the best sequenced ever!

natedey, 2022-11-02:
It would be cool if once in the GEN menu, you could just pick "DICEROLL" or something, which would randomize seed/smooth/bias/scale and give you results with many less clicks.

Mauv3, 2023-01-08:
It would be a great addition but I think that having control on smooth, bias and scale is helpful in order to achieve more musical results.

natedey, 2023-01-08:
why not both? :)